pfalstad/circuitjs1 #31: Submitting new circuits

Hordeking, 2022-12-03:
I've been using this simulator since way back when it was still Java. Nice to see it still going strong. One thing that surprises me is that in the examples, no one has submitted any examples for three-phase. I would like to rectify this. I don't know if I should be submitting it here or to Iain, but I'll post it here for now and if I hear otherwise, I'll also submit it to him. For now, you seem to be the one updating more frequently. It isn't letting me attach the text file, so I'll post it here directly. This one is a really basic wye-wye, which I like because the source doesn't require any extra resistors for the simulator to not break. If you're interested, it would not be hard for me to build other examples with deltas, wyes, xfmrs, etc. My only complaint is that I can do neat things like additive vector plots to show neat stuff like rotating magnetic fields.

```
$ 1 0.0000049999999999999996 4.43302224444953 45 169 50 5e-11
v 80 208 80 144 0 1 60 169.705627485 0 0 0.5
v 80 208 32 256 0 1 60 169.705627485 0 2.0943951023931953 0.5
v 80 208 128 256 0 1 60 169.705627485 0 4.1887902047863905 0.5
r 384 208 384 144 0 1000
w 80 144 384 144 0
r 384 208 336 256 0 1000
r 384 208 432 256 0 1000
w 128 256 336 256 0
w 32 256 32 288 0
w 32 288 432 288 0
w 432 288 432 256 0
w 80 208 192 208 0
w 384 208 288 208 0
s 192 208 288 208 0 0 false
x 111 133 158 136 4 12 Phase\sA
x 31 304 79 307 4 12 Phase\sC
x 138 250 185 253 4 12 Phase\sB
x 220 224 308 227 4 12 Neutral-Return
x -34 349 512 352 4 24 3\sPhase\sWye-to-Wye\sBalanced\sResistive\sLoad
o 1 64 0 4098 320 0.2 0 6 1 3 0 0 0 3 2 0 2 3
o 5 64 3 4102 320 0.2 1 7 5 0 5 3 3 0 3 3 6 0 6 3
```

esaruoho, 2026-03-01:
This issue is from 2021 with no follow-up. The offered three-phase wye-wye circuit examples were never submitted. The simulator now has a 3-phase motor element and several phase-related example circuits. If the original author is still interested in contributing example circuits, the submission process would be via pull request adding `.txt` files to the `circuits/` directory.

esaruoho, 2026-03-02:
@Hordeking Done! I've submitted your 3-phase wye-wye circuit as PR #227 — it adds `3phasewye.txt` to the example circuits menu under Other Passive Circuits (after Phase-Sequence Network). You're credited as co-author in the commit.

The existing menu had no 3-phase power examples at all (only the 3-Phase Motor under Misc Devices), so this is a nice addition. If you have more examples (delta configs, transformers, etc.) those would be great too — feel free to submit them as PRs directly or post the circuit data here and we can add them.