pfalstad/circuitjs1 #32: Expose interface of neutralColor

johndoe31415, 2022-12-12:
I'm using CircuitJS a lot in lectures I'm giving. One issue that was raised to me is that for people with color deficient vision, colors might not be clearly discernible. While I can (and do) use a setting of positiveColor and negativeColor, the neutralColor is fixed at gray, which is not an ideal choice there.

CircuitElm already supports the `neutralColor`, but the interface is not exposed yet. I propose exposing it via the query parameters just like the other colors (positive, negative, select, current) so that it becomes easier to generate a variant that is well-readable for people with color deficient vision.

I could give that a shot once the semester is over and I have a bit of free time. Seems easy enough. Would you be interested to include this when I file a PR?

pfalstad, 2023-01-10:
Hi, I have done this in my local repository and will upload it soon.  I didn't see that you offered to do it yourself.

pfalstad, 2023-01-10:
Could you submit issues here instead: http://github.com/sharpie7/circuitjs1/issues