pfalstad/circuitjs1 #36: Non-inverting op amp phase incorrect

wcalvert, 2023-01-03:
Hello, I'm attaching the text that illustrates a non-inverting op amp circuit with a phase shift of 180* degrees on the first scope plot, and a correct in-phase non-inverting op amp circuit on the second scope plot. It seems that the op amp model with the negative terminal on top consistently causes this issue. Thanks for your consideration.

Platform is Chrome 108.0.5359.125, Windows 10.

$ 1 0.000005 30.13683688681966 50 5 50 5e-11
w 528 128 528 192 0
w 432 128 432 176 0
g 432 64 432 32 0 0
r 432 128 432 64 0 10000
r 432 128 528 128 0 10000
a 432 192 528 192 8 15 -15 1000000 3.1432826843313153 3.1433455499850016 100000
w 432 208 368 208 0
v 368 208 368 272 0 1 40 5 0 0 0.5
g 368 272 368 288 0 0
g 848 256 848 272 0 0
w 848 48 848 112 0
w 752 48 848 48 0
r 848 256 848 192 0 10000
r 848 192 992 192 0 10000
w 992 128 992 192 0
a 848 128 992 128 9 15 -15 1000000 -3.143282684331315 -3.1433455499850016 100000
w 848 144 848 192 0
g 752 128 752 160 0 0
v 752 128 752 48 0 1 40 5 0 0 0.5
o 7 64 0 20482 10 0.1 0 4 7 3 0 0 0 3 Bugged
o 11 64 0 20482 10 0.1 1 4 11 3 14 0 14 3 Correct

pfalstad, 2023-01-03:
You just need to swap terminals in the first AC source.  The * indicates the positive terminal.  In the circuit on the left, the * terminal is attached to ground, which is not right.

wcalvert, 2023-01-03:
I hadn't noticed that, thank you for the quick response.

closed 2023-01-03 by wcalvert