pfalstad/circuitjs1 #46: couple of questions to see if this is possible.

kdschlosser, 2023-04-07:
I work with MCUs like the ESP32, STM32 and NXP series. I mostly work in MicroPython as it is easier to prototype because of it compiling at runtime. One of the things that I have not come across is a simulator that is geared in a more advanced manner. I have found ones that use Blockly which is geared more for the newbie.

I work with LVGL graphics library quite a bit and they have put together a means of compiling their software with MicroPython and because MicroPython has the ability to be compiled into a javascript they have put together a pretty decent simulator but it is missing the ability to add functional circuits.

You can check out the simulator they have here.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/widgets/slider/lv_example_slider_2.py&script_direct=0d22c3378d4bb43ac657c93bd01f17d21ac6bde1

I know your circuit simulator can be programmatically altered using websockets and I am interested in knowing more about the functionality.

This is what I am thinking. The user would be able to select an MCU they want to develop for from a list. This would then launch MicroPython with the proper modules loaded into it. On startup it would make a websocket connection to the circuit simulator and pass information to it about the MCU. The user would be able to run MicroPython code that would be the same as if they had the code loaded onto an actual MCU but instead it would be talking to the circuit simulator to get or set GPIO states.

Is that something that is possible?

pfalstad, 2023-04-09:
Hi, the websocket interface was contributed by someone so I know little about it, other than what you can read here: https://github.com/pfalstad/circuitjs1/tree/master/websocket

You can load circuits and set voltage sources so that might be enough.  I'm not sure what MCU information you need to pass to the simulator.

kdschlosser, 2023-04-10:
I need to be able to tell the simulator when there is a voltage coming out of a GPIO from the MCU to activate a circuit int the simulator. I also need to be able to collect voltages from a circuit connected to a GPIO in a circuit.

I would need to be able to add an MCU to the simulator as well where the GPIOs would be defined so a person could draw out the circuits and connect them to the GPIOs

BBBSnowball, 2023-04-29:
I might be pointing out obvious things here... If so, please feel free to tell me and I'm sorry in advance ;)

There is a version with an [AVR8 simulator](http://www.falstad.com/circuit/avr8js/). I think this is doing what pfalstad described above, i.e. run the MCU simulator in parallel to circuitjs and update "external voltage sources" in the circuit, which represent the MCU pins. This approach should work with any MCU, as long as you only need one of them at a time. The upside is that you can easily have dedicated UI for the MCU, i.e. a code editor.

The other option would be to make a custom circuit element, e.g. based on `ChipElm`, as described in [INTERNALS.md](https://github.com/pfalstad/circuitjs1/blob/master/INTERNALS.md#adding-new-elements).

(I'm looking through the existing issues because I'm interested in making a custom circuit element that can be programmed in Javascript. If I get this into a usable state, this might be another option - but I think the existing options are better for your case.)

kdschlosser, 2023-05-01:
This gives you some idea of what I am working with.

https://www.youtube.com/watch?v=j9NI1cFOKrE

All bundled together, code editor, display to simulate graphics output from LVGL, the last missing piece is the circuit simulator with the ability to add pins and tell the circuit simulator what the voltage is if there is output from a pin, or reading voltage from the circuit simulator for a specific pin.

What you see in that video is aa port of Python called MicroPython. It is able to be compiled and run as Javascript. This specific port has been compiled with a GUI framework, hence why you see the animated graphic in the upper right. The code editor is right there and a simple click of the button runs any code in the editor instantly. If there was the circuit simulator it would be a really nice way to develop for an MCU.

MicroPython is able to be compiled for a bunch of different MCUs and also things that are not an MCU as well. It can also be compiled to run on Windows, Linux and also Javascript. LVGL which is the graphics library can run on just about any kind if a device so long as it meets these minimum specs:

16bit processor
16MHz
64kB of program storage
8kB of RAM
and is compiled using C99 or newer.

I just wrote a binding to LVGL for CPython. You can use any thing that has the ability to draw a frame buffer to the screen. There are 2 examples with it using SDL2 and wxPython. The binding is still in the early stages of development but it is mostly functional. That binding will make it possible for people to use the IDE/simulator to develop for a Raspberry Pi which can run CPython

The link that @BBBSnowball posted is along the same lines. Add the display support and remove the need to compile the script server side. MicroPython scripts are just like normal Python scripts they are compiled at runtime. because MicroPython is running using JavaScript on the local machine that runtime compilation of the script gets done on the local machine. The script doesn't even get sent to the server to run it.

kdschlosser, 2023-05-01:
I also did want to mention that the display portion is fully interactive and accepts user input.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/widgets/slider/lv_example_slider_2.py&script_direct=a8f71d4c35848dbf796f35e126a5361ad622cbd1