pfalstad/circuitjs1 #49: Decoupling capacitor causes error message: "Capacitor loop with no resistance!"

FoobarTheFatPenguin, 2023-04-22:
If I put a capacitor between positive and negative of a DC voltage source, I cannot start the simulation. It keeps complaining that there's a capacitor loop without a resistor. But decoupling caps don't need a series resistor, as far as I know. Granted, I'm not an electrical engineer, but every example I can find on the web always has them directly between + and - (which makes sense to me).

Example circuit (I use a simple resistor as a load, but imagine it's something that requires a stable voltage, like a µC):

```
$ 1 0.000005 10.20027730826997 50 5 43 5e-11
v -352 96 -352 -64 0 0 40 5 0 0 0.5
w -352 -64 -288 -64 0
w -352 96 -288 96 0
c -288 -64 -288 96 0 0.00001 0 0.001
w -288 -64 -224 -64 0
w -288 96 -224 96 0
r -224 -64 -224 96 0 1000
```

I apologise if my lack of electronic knowledge is the issue here, but seems to me that this circuit should work. I actually built it on a breadboard, and the universe didn't collapse. ;)

pfalstad, 2023-04-22:
The circuit will work.  It's just a limitation of the simulator.  See here: https://github.com/sharpie7/circuitjs1/issues/634

For now just add a 1 milliohm resistor in series.

FoobarTheFatPenguin, 2023-04-22:
I see. Thank you for your reply. I'll do the workaround then.

closed 2023-04-22 by FoobarTheFatPenguin