pfalstad/circuitjs1 #51: Ctrl + 2 fingers click = 2 fingers click = context menu in the way on Mac M1

dikkietrom, 2023-05-13:
with the mouse it would be ctrl + left click for dragging pops up the context menu that is normally under right click. Because the context menu is over the point you want to drag, it does not work or very inconvenient with many try and error.

pfalstad, 2023-05-14:
I don't understand.  You're saying that ctrl + left click brings up the context menu, and you don't want that to happen?  Why are you holding down ctrl?

dikkietrom, 2023-05-18:
the behaviour is different but that is maybe a consequence of the context popup. With ctrl I can drag one end only. With shift it drags a junction. 
[image]
vs
[image]