pfalstad/circuitjs1 #52: Zooming and panning doesn't work on mobile

mmghv, 2023-05-15:
I'm unable to zoom or pan with two fingers on mobile, I'm able to zoom and pan on the side panel or if I open any menu like long press to open the add menu, but not on the circuit itself.

https://github.com/pfalstad/circuitjs1/assets/5418859/13381089-1835-4717-9bf9-5af7b4eab253

Also I've noticed that I lose the ability to edit the circuit once I start zooming or panning, when I reload the page I'm able to edit, when I start to zoom, the screen shrinks and I lose the ability to edit.

https://github.com/pfalstad/circuitjs1/assets/5418859/6728d488-c40c-46a4-8bb5-c14b3f12485f

Tested on both Chrome & Firefox
Android 12
Xiaomi Redmi Note 9 Pro

pfalstad, 2023-05-15:
Yes, this is an issue with android, because I used a iOS-only feature when implementing it.

mmghv, 2023-05-16:
I checked and now it's working perfectly, Thank you for the quick fix!

closed 2023-05-16 by mmghv