pfalstad/circuitjs1 #59: 555 timer with non-ground negative supply

azambon, 2023-09-16:
Hello,
I stumbled across an issue with the 555 timer. It appears that if the "gnd" pin is connected to a voltage other than 0V, the component misbehaves.
Take a look at the attached screenshot: two oscillators with the exact same components and configuration and both supplied with 5V, except that one uses a positive supply (+5V / 0V) while the other uses a negative supply (0V / -5V). One would expect both circuits to behave the same, except that the voltage levels are shifted. Instead the one with negative voltages does weird things.
[image]

Here's the link to that sample circuit: https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3EwwEx2QdnQZkgDmQGwCcR6IArJBRVeQKYC0YYAUGAeSACxe4ip8uBLvzwgC3aCWklK8BCwDG3diCxYJXVVzhRwjMmGjZIwyFlyR0RJDzIwFAJxUT2VLROQ69XdPBYA7hTont7kRMj83pCBFBFRtPFuULHhkcnkYOkEVDFBmdm0WeAhKfnFyR7gOSnOVV7uqjgiVEhwsTx8DS4JZdy8apAiVc19I0M96hJ5kxr9fFN95DxqBHzLw7ozG9y6VTq5sTaRByDouJuHQcerfDenM+cii7hNc48XgyJPau+xrxJRj9RjMAb0ft0Ys4Id5gRNWv4grhLNUqD9kqCUclka1SjMiDVsViaviat0Ca0SUcVuSydEWNDPskYbluH52spgXMfg99AwyFxoFlwuYQgQwBpkEROA5IKx2JwwQIQLgsJFldMWBp1qUMJEcl10JFDtqQEQaUazbrLYcAEo9SqqLJ8VlhPS5aDkFj29E1dHIF0+VnBz3enqnfZq927aMer0+z6nHlR11UBgy92hgD2gxVQaI63E0Ba-FzWBYOawEEDWSl7ikMlk8gQUFgLaokSrahYQA

azambon, 2023-09-17:
I had a chance to take a look at the code.
[Here](https://github.com/pfalstad/circuitjs1/blob/c192b71e325624e56d7ed9db71456b48d2e48871/src/com/lushprojects/circuitjs1/client/TimerElm.java#L97) is a possible issue, in that the check `volts[N_OUT] > volts[N_VIN]/2` does not seem to take into account that the ground pin may be connected to something other than ground. It should probably become something like `volts[N_OUT] > (volts[N_VIN]+groundVolts)/2` after moving the definition of groundVolts to the top of the method.
Same [here](https://github.com/pfalstad/circuitjs1/blob/c192b71e325624e56d7ed9db71456b48d2e48871/src/com/lushprojects/circuitjs1/client/TimerElm.java#L104), where the `volts[N_CTL]/2` expression does not yield the expected value in case of a non-ground negative supply. `(volts[N_CTL]+groundVolts)/2` should fix it, I think.

I cannot find anything else in that file, so perhaps fixing those two items would be enough?

pfalstad, 2023-09-17:
yes, I think that is right.

pfalstad, 2024-03-15:
fixed

closed 2024-03-15 by pfalstad

pfalstad, 2024-09-13:
I don't believe so.  I got that circuit from here:

https://www.electronics-tutorials.ws/waveforms/555_oscillator.html

Scroll down to "50% Duty Cycle Astable Oscillator".  That circuit also has the discharge pin unconnected.