pfalstad/circuitjs1 #60: Apple Silicon (ARM) version ?

John-Lluch, 2023-10-11:
Do you plan to provide a Mac version for Apple Silicon (ARM) ?

The app currently runs under Rosetta.

Thanks in advance !

pfalstad, 2023-10-11:
Hi, try this:

https://www.falstad.com/circuit/offline/CircuitJS1-macarm.dmg

The offline versions are just Electron with the circuitjs javascript code included.  Electron supports Apple Silicon so this should work.

John-Lluch, 2023-10-12:
Thanks for the quick answer. I downloaded it and an app named Electron.app appears inside the dmg. However when attempting to open the app the message "Electron.app is damaged and can't be opened" . I'm not sure if I am doing something wrong of there's an issue on the app download?

John-Lluch, 2023-10-12:
Never mind... I fixed that by running the following Terminal command

`xattr -cr /Applications/Electron.app` 

The app is now more responsible and runs smoothly on my Mac. Thanks again !!

pfalstad, 2024-03-06:
The ARM version should work now without xattr.

closed 2024-03-06 by pfalstad

gabemorris12, 2025-01-12:
I can't get the arm version to work. I get the damaged error like the OP, but when I run `xattr -cr /Volumes/CircuitJS1/CircuitJS1.app`, then try to open it, it still won't work.