pfalstad/circuitjs1 #63: Heavy performance degradation in v2.8.1 compared to v2.7.4

TishSerg, 2023-11-17:
Same circuit, same circumstances, same settings, but heavy difference:

2.8.1:
[image]

2.7.4:
[image]

pfalstad, 2023-11-17:
It's not slower, it's actually slightly faster!  In 2.8.0, we have a minimum framerate of 20 fps.  In 2.8.1 I lowered the minimum to 2 fps.  According to my test, the simulation runs faster but the display is much choppier.  That's because we're running the simulation longer before stopping to update the display.  Someone requested this (#54).  Reduce the simulation speed to make the display less choppy.

TishSerg, 2023-11-17:
> Reduce the simulation speed to make the display less choppy.

That did the trick. Thank you.

> In 2.8.0, we have a minimum framerate of 20 fps. In 2.8.1 I lowered the minimum to 2 fps.

How about making this adjustable? Say in "Other Options"...

pfalstad, 2023-11-18:
probably a good idea to make it adjustable.  2 fps is too low.

pfalstad, 2024-03-15:
It's adjustable now and is higher by default.

closed 2024-03-15 by pfalstad