pfalstad/circuitjs1 #70: Coil voltage on open relay contacts ?

rolox2022, 2024-02-07:
Hi, i noticed a strange behaviour:
there seems to be some "interconnection" between coil and contacts of a relay .... 
can be reproduced with simple circuit attached.
[image]
[relay issue.txt](https://github.com/pfalstad/circuitjs1/files/14199548/relay.issue.txt)

rolox2022, 2024-02-11:
Edit...its obviously a misunderstanding on my side...the parameter "off resistance" is by default 1M which affects low current signals switched by relay and also is related to contact-contact resistance as kind of "isolation resistance"
As in the example no current is applicable with only testpoints, the full 5V show up at the open contacts ....

pfalstad, 2024-02-11:
Yes, that's correct.  You need a pulldown resistor if you want the open contact to be grounded.

rolox2022, 2026-03-01:
Hi, bit older ...and solved for my me.. thx...