pfalstad/circuitjs1 #76: feature: analog multiplexer

conversy, 2024-03-06:
Hello,

the currently available Multiplexer is digital-only: the output is the "High Logic Voltage" if the selected input pin's value is True.
It would be nice to have an analog option, that would basically transfer the selected input pin's voltage/current to the output.
May be something like this in `MultiplexerElm.java`, though I'm not sure how the internals of CircuitJS work:

```
class MultiplexerElm extends ChipElm {
	final int FLAG_INVERTED_OUTPUT = 1<<1;
	final int FLAG_STROBE = 1<<2;
	final int FLAG_ANALOG = 1<<3;
...
	int getSelectedValue() {
	    int selectedValue=0;
	    for (int i = 0; i != selectBitCount; i++)
		if (pins[outputCount+i].value)
		    selectedValue |= 1<<i;
		return selectedValue
	}

	void execute() {
	    int selectedValue = getSelectedValue();
	    if (hasFlag(FLAG_ANALOG)) {
		double current = pins[selectedValue].current;
		if (strobe != -1 && pins[strobe].value)
		    current = 0;
		pins[outputPin].current = current;
	    } else {
		boolean val = pins[selectedValue].value;
		if (strobe != -1 && pins[strobe].value)
		    val = false;
		pins[outputPin].value = val;
		if (hasFlag(FLAG_INVERTED_OUTPUT))
		    pins[outputPin+1].value = !val;
	   }
	}
```

I hope this makes sense!
(And congrats for this excellent software!)

conversy, 2024-03-06:
Thank you for answering!

Yes, it would be directional, I guess.
I read somewhere that such chip has an internal resistance of about 60 ohms, for what it's worth.

AgainPsychoX, 2024-03-25:
Anyway could it be compatible with something like [`74HC4052`](https://www.sparkfun.com/datasheets/Components/General/74HC4052.pdf)? Maybe I can use sub-circuit to simulate something like this, using MOSFETS?

AgainPsychoX, 2024-03-26:
Indeed I managed to make it with sub-circuits, Or at least I think I do, it seems to working for me, please notify me if something is wrong. 😄 

Here circuit files (2 versions, as logic level for HIGH is depending on VCC):
[74HC4052_3_3V.txt](https://github.com/pfalstad/circuitjs1/files/14751035/74HC4052_3_3V.txt)
[74HCT4052_5V.txt](https://github.com/pfalstad/circuitjs1/files/14751037/74HCT4052_5V.txt)

You can `File > Open File...` and then `File > Create Subcircuit...` and then move the outputs around. 

+ There is VCC that should be linked to specific voltage (3.3V/5V), or you can join it before importing and forget about it. The same goes for GND label, it's there purely to allow it look like DIP16 on the schema. They could be removed, but it would require to remove GND and mark every VCC label as internal node and join it to 5V.

BTW I like CircuitJS so far, there are few annoying things, but it's huge help.

AgainPsychoX, 2024-03-26:
Actually there seems to be error in the design, it doesn't work with negative VEE, which is should, and on top of that voltages are weird at best. I will investigate, try to fix it and update the design. I am not really good at electronics :/

conversy, 2024-04-26:
(sorry for the delay)

@AgainPsychoX thank you so much!

However, I'm not quite sure I understood how to use subcircuit in circuitjs, even after searching through "directions" on the falstad.com site... How to add one of your circuit into my own and use it?

AgainPsychoX, 2024-04-28:
(Sorry for late response, I missed the e-mail).

Here are instructions:

1. Open the app in new tab https://www.falstad.com/circuit/circuitjs.html
2. From the menu in right top corner: `File` > `Open File...` (or use `Ctrl`+`O` keyboard shortcut).
3. Select the circuit file you downloaded - there are multiple versions, here I will go with [74HC4052_5V_SIMPLE.txt](https://github.com/pfalstad/circuitjs1/files/14765728/74HC4052_5V_SIMPLE.txt).
4. The circuit contents should be loaded and displayed. From the menu: `File` > `Create Subcircuit...`:
[image]
5. Now you have to model the reusable element that will be representing the subcircuit. 
    + Fill the model name so you can later find it. The name will also be displayed on the element if you select the `Show label` checkbox. I will go here with `74HC4052_5V_S`. 
    + You can leave pins as is, or you can move them around by pressing and dragging around. In my case, I rearranged the layout the pins to match DIP16 (flipped X&Y) because it was easiest to drop in to a simulation for a project I was working on.
    + Select the `Save Across Sessions` checkbox, so the subcircuit is available even in other tabs, after refresh and after browser restarts.
    + After you are finish, press "OK". 
[image]
6. You can close the tab and/or open blank circuit, or any other target circuit where you want to place it. 
7. To place the subcircuit, use from the menu: `Draw` > `Subcircuits` > `Add <name you choosen>`:
[image]
8. Click anywhere on your circuit (or if doesn't work: press and drag a bit, it's weird sometimes). After placing the instance, you might want to exit from the drawing mode by pressing `Esc` or `Space` or using the menu: `Draw` > `Select/Drag Sel`. 

That's should be it. Extra notes: 

+ You can edit the subcircuit by right click menu > `Edit...` (or double clicking). Care: `Load Model Circuit` might replace your current circuit to display the contents; also it might be unavailable if you try edit circuit used inside other circuit.
[image]
+ The subcircuits are saved in your browser local storage (if you selected the `Save Across Sessions` checkbox). 
+ To delete the subcircuits, you can open the website, and open browser developer console (Chrome/FireFox keyboard shortcut is `F12`), then navigate to `Application` (Chrome) or `Data` tab (Firefox). After selecting the website under `Storage` > `Local Storage` you can see all the data the website saves in your browser storage, including subcircuits with keys prefixed with `subcircuit:`, in my case:
[image]

Let me know if anything is unclear, I will try to help if I can.

conversy, 2024-06-04:
@AgainPsychoX Sorry I saw your answer the other day but could not find the time to implement it... Thank you so much, these are very clear explanations, that might deserve being put somewhere on the circuitjs website, so that others can find them easily...

Thanks to your explanation, I could create the part in a subcircuit and use it in another circuit.

Your design is not exactly what I expected (it seems there are two input, while I only have one in the part I want to simulate), but I can adapt yours easily I think.

Many thanks again!