pfalstad/circuitjs1 #77: Abnormal BJT current source behaviour with oscillating transformer in circuit

NE558P, 2024-03-13:
Hello,

Week ago or so I've discovered weird behaviour of transistors while playing with circuit known also as joule thief. I wanted to experiment a bit with it and added simple current source made from two PNP transistors and three resistors. While joule thief part is not oscillating, current source works as expected, otherwise it just starts giving abnormal high current pulses even, if I disconnect it from joule thief part and switch back to resistor. I though problem was with web browser (JS engine?), but this happens on MS Edge and Firefox.
I know this simulator is not best to do such things, but it's still very useful and I like to play with it a lot.

I'm attaching screenshots and source file for reproduction:

[image]
[image]
[image]
[fail_src.txt](https://github.com/pfalstad/circuitjs1/files/14590025/fail_src.txt)

Greetings,
Sebastian.

azambon, 2024-03-15:
@NE558P Not an expert, but those 1k resistors in series with the transistor bases look suspicious, especially the one in series with the base of the leftmost transistor.
Does the problem persist if you remove them?

NE558P, 2024-03-15:
> @NE558P Not an expert, but those 1k resistors in series with the transistor bases look suspicious, especially the one in series with the base of the leftmost transistor. Does the problem persist if you remove them?

Yes. Problem still exist whether you have base resistors or not:
[image]

pfalstad, 2024-03-15:
We discussed this in email but I'll just repeat it here: there is a clear bug because the current source on the left half of the circuit has current spikes due to the behavior of the right half of the circuit, even if they are disconnected.  It seems likely to be a roundoff error issue.