pfalstad/circuitjs1 #84: Mouse Scroll Wheel Resistors and Capacitors Values

jjdnd, 2024-04-21:
please can you disable the Mouse Wheel adjusting Resistor and Capacitor Values? it is extremely annoying when you try to zoom in and out and the scroll wheel to zoom in and out randomly adjusts circuit values ruining your work if the mouse happens to be in range of a component

pfalstad, 2024-04-22:
there's an option to turn it off, under the Options menu

BETLOG, 2024-04-25:
I'm seeing the opposite problem: when i intentionally try to use the scrolling feature it doesnt scroll the numbers at all.

pfalstad, 2024-05-04:
@BETLOG That's strange, the scrolling feature never works for you?  What platform are you on?  Offline or on the website?  What browser?

BETLOG, 2024-05-04:
Never scrolls the numeric options after i middle click and the scrollmenu appears, and disallows clicking anything.
Kubuntu 23.10
Linux  6.5.0-27-generic #28-Ubuntu SMP PREEMPT_DYNAMIC Thu Mar  7 18:21:00 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Google Chrome Stable Version 123.0.6312.105 (Official Build) (64-bit)
https://falstad.com/circuit/circuitjs.html

pfalstad, 2024-05-05:
Middle click doesn't do anything at all for me.  Do you have a scroll wheel?  Scrolling up and down allows you to select a value.  Clicking on them doesn't do anything.

jjdnd, 2024-05-05:
Middle click doesn't do anything for me ether, im just glad to find i can disable it
Ubuntu 22.04.4 LTS
Firefox 125.0.2 (64-bit)

BETLOG, 2024-05-05:
> Do you have a scroll wheel?

Sorry, yes, thats what I mean.
Middle click (the scroll wheel pressed into the mouse) does nothing.
Scrolling: brings up the menu, but does not scroll through the values. Left, middle, or right  clicking any value dismisses the menu but does not set any value selected.
So the menu appears when scrolled over, but input of any kind is impossible.

mouse:
Logitech G502 HERO (wired)
Logitech G602 mouse (wireless)
both give 'identical' results:
```
$ xev -event button # then scroll down one notch, up one notch
Outer window is 0x5600001, inner window is 0x5600002

ButtonPress event, serial 25, synthetic NO, window 0x5600001,
    root 0x1e3, subw 0x0, time 1500027, (52,90), root:(923,589),
    state 0x0, button 5, same_screen YES

ButtonRelease event, serial 25, synthetic NO, window 0x5600001,
    root 0x1e3, subw 0x0, time 1500027, (52,90), root:(923,589),
    state 0x1000, button 5, same_screen YES

ButtonPress event, serial 25, synthetic NO, window 0x5600001,
    root 0x1e3, subw 0x0, time 1501855, (52,90), root:(923,589),
    state 0x0, button 4, same_screen YES

ButtonRelease event, serial 25, synthetic NO, window 0x5600001,
    root 0x1e3, subw 0x0, time 1501855, (52,90), root:(923,589),
    state 0x800, button 4, same_screen YES

```

pfalstad, 2024-05-05:
try scrolling in one direction much further than you think is reasonable and see if the value changes.

jjdnd, 2024-05-05:
the option to disable editing values with scroll wheel is no use. it also disable zooming the graphs
but still allows zooming the schematic
the graph is not a value that has any affect on editing the circuit design

@BETLOG i wish mine did that, it is so annoying that zooming in and out ruins your work :-)

pfalstad, 2024-05-05:
@jjdnd hmm, that's a bug

BETLOG, 2024-05-06:
> try scrolling

100 upward scrolls of ~2 notches each (it isnt a smooth scrollwheel, it has tactile indents for each activation) 
...nothing

> zooming the graphs 

I don't have this either.

> @BETLOG i wish mine did that, it is so annoying that zooming in and out ruins your work :-)

I've never had that problem; but: agreed. It should really be an option for key combination only; like ctrl-scroll or shift-scroll before it changes anything.

Also perhaps related, and definitely undesirable: when i scroll the general schematic to zoom it is extremely slow, each scroll is only a tiny zoom change, and somewhat tedious to even frame the entire schematic, i would never use it to momentarily zoom on a section of circuit. At best its only useful for maximizing the circuit in the screen and leaving it at that. Panning works as expected though.

pfalstad, 2024-05-06:
That sounds very related.  Someone else on linux was saying that after a change I made to the scrolling, it takes a lot of steps to make the wheel turn.  It's related to the device pixel ratio (the scaling).  Maybe your scaling is so high that it takes an infinite number of steps.  (We use integer division to calculate how many steps to scroll the wheel.)

This is https://github.com/sharpie7/circuitjs1/issues/808

BETLOG, 2024-05-06:
> That sounds very related.
> This is [sharpie7#808](https://github.com/sharpie7/circuitjs1/issues/808)

Sounds, yes... however my scale is set to 100%, and afaik all other related settings are default. Kubuntu didnt have this 'display percent' setting back when I started using it, so I habitually set DPI and font size instead.

[image]

...But evidently I dont even need to change DPI on this screen/OS now either.
[image]

Even my mouse settings only differ from default in that I use flat/no acceleration.
[image]

It may be worth mentioning that I use evrouter and xmodmap for binding mouse buttons to desired keys, but none of those binds include the scrollwheel. The xmodmap aspect is only engaged when I'm playing a game and reset to default afterward, and even on a fresh boot with evrouter killed I still have the circuitjs.html-component-scroll issue.

I'm grasping for possible connections to the issue now.

```
$ xset -q
Keyboard Control:
  auto repeat:  on    key click percent:  0    LED mask:  00000000
  XKB indicators:
    00: Caps Lock:   off    01: Num Lock:    off    02: Scroll Lock: off
    03: Compose:     off    04: Kana:        off    05: Sleep:       off
    06: Suspend:     off    07: Mute:        off    08: Misc:        off
    09: Mail:        off    10: Charging:    off    11: Shift Lock:  off
    12: Group 2:     off    13: Mouse Keys:  off
  auto repeat delay:  600    repeat rate:  25
  auto repeating keys:  00ffffffdffffbbf
                        fadfffefffedffff
                        9fffffffffffffff
                        fff7ffffffffffff
  bell percent:  50    bell pitch:  400    bell duration:  100
Pointer Control:
  acceleration:  2/1    threshold:  4
Screen Saver:
  prefer blanking:  yes    allow exposures:  yes
  timeout:  0    cycle:  600
Colors:
  default colormap:  0x20    BlackPixel:  0x0    WhitePixel:  0xffffff
Font Path:
  /usr/share/fonts/X11/misc,/usr/share/fonts/X11/Type1,built-ins
DPMS (Energy Star):
  Server does not have the DPMS Extension

```

Can anyone confirm if this issue exists on 24.04 *buntus?