pfalstad/circuitjs1 #85: 'show frequency' is intermittent

BETLOG, 2024-04-25:
Whenever I scope something with a waveform faster than ~30KHz it often stops displaying the frequency as text.
Other times it will display frequencies ABOVE ~35KHz.

Adding audio sources with 48KHz sample rates triggers the option to increase horizontal scale resolution, which is sometimes helpful, but I'm not sure if the value propagates to existing scopes.

> options>other options> Auto-Adjust Timestep / Minimum time step size (s)

Doesnt seem to help, and I'm not sure if its setting always propagates either.

I'm an electronics noob who is trying to see what I remember, so I tend to work around the intermittent frequency display, and havent done any empirical  testing on the problem, so maybe its just me, but it feels like an intermittent bug.

[image]

pfalstad, 2024-05-04:
Hi, when you post a circuit, could you post a link (File->Export As Link) rather than a screenshot.

The frequency doesn't show up steadily if the simulator is having trouble figuring out the frequency.  To help with this, adjust the scale and/or make the scope bigger.

BETLOG, 2024-05-04:
> adjust the scale and/or make the scope bigger.

Yeah, i tried a few things, adding oudio and maxing its sample rate seems to help, but there is almost always a dead spot when i hit 30KHz

https://falstad.com/circuit/circuitjs.html?ctz=CQAgbArCAMB00IQJjNALARjNnvsgGYAOJWAdjCQhIzQIE57ow1wp3oQIBTAWgwwAoLFExgQWTmhYTUE1hnK0CaItCRE0TMASQE+ZQQHcuGJCDEW1FrDEEAnQpon1zxVhqIx5EJMa5oCq4BChp2JhCBEmGqnGZe0P6x0V7uKeE24p6ZIFTiiRFmOahelokAbrlEpbbZed5SnPQNMLAQggAmVV712fG5IB3cAGYAhgCuADYALv4EBOL9af0FhAs5aWWCSNDN84th+zmccIgIjBeXV4wgRLCYSBqoZLRg9BDrp9AYEv7ZlkctiY+od1isHE4gm5nCsLDs5s5LMkgTlsmlsqtLBhgstgokAMZWTi6KTWMoSPheCCwEpgMj0MjQapaMjUVoIISFNw7UzmLaE9HVSHHCm8VKwIgELAYBZgMwyhhIVhfTkhdIldKJRwajAUcBM3JK7yUBAQjXZc1GuI7VbmoUarYAB3AZHMj1YdL5tgIDX8nuKBq2XUiHiNlo8gxGExmnX1Xn6Or15iGYymsxM4Zdbqt2wEWcNrEZCiicSJCF+AHtcotvBB+hgTrAfnFBFWUBJvJgGxJG83fsQBgAxPsy8u68v8EAAJW4AGcAJaz6ajAB2+O4giAA

pfalstad, 2024-05-05:
Reducing the timestep will help and will also make the simulation more accurate.  See Options->Other Options and reduce it to maybe 0.5u .

closed 2024-05-06 by BETLOG