pfalstad/circuitjs1 #86: Scope Vertical Scale

jjdnd, 2024-04-28:
I cannot find any setting anywhere to change Scope Vertical Scale divisions from 8 to 10,
Im sure there is a logical reason why 8 divisions is useful to someone but dividing any base 10 number like 1 volt into peace's would be much more useful to me as whole numbers or single decimal placed numbers in this instance, 
like 0.1v 0.2v 0.3v NOT 0.125v 0.25v 0.375v

pfalstad, 2024-05-04:
Hi I'm not sure what you mean, could you post a screenshot?

jjdnd, 2024-05-04:
2 screenshots attached

the group has horizontal lines separative the vertical into in this example 8 divisions

it appears to double the divisions at a set graph height, as in drag the graph taller and every so often the number of divisions double, still a multiple of 8

why 8?

then try and manual position a plot at the bottom of the graph, the CH1 Position slide bar has no numbers of any kind for reference, how am i supposed to set CH2 at the same number since i don't have a clue what the number is (notice the red and green lines are out of position from each other and probably out of line with the horizontal lines as well)

finally CH3 yellow cannot be removed from the scope properties, if you can hover the mouse over it on the graph then you can remove the plot, but since it defaults to always being under CH1 or CH2 there is no way to get rid of it without manually positioning it out of the way.

if CH3 was not there, the auto positioning would work better, its not very good when there are both V and I on the same graph. being able to remove CH3 would likely eliminating the need to set it manually anyway but trying to remove plots you don't want is difficult, it would be nice to have a plot remove option in the Scope Properties, and numbers relating to the position slider like there are for scale. 

Scale V/div 
is div a division? as in the divisions i am trying to explain are 8 in total?

[image]
[image]

pfalstad, 2024-11-29:
Hi, to get rid of CH3, uncheck "Show Current".
Working on an adjustable number of vertical divisions.