pfalstad/circuitjs1 #87: Labeled Nodes

jjdnd, 2024-05-02:
what is the difference between the labelled nodes in the following link?

look at the nodes labelled out on the right hand side, all the the nodes have a slightly different font size and manipulating and placing them behave differently and fail to snap properly to the grid.

this usually happens if you copy and paste circuits from the circuit menu at the top or from within your Subcircuits examples on the top menu, the labelled nodes appear to be different from the labelled nodes in the right click and labelled node option

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCAMB0l3BWEBGGAmOaDsWyQBxoBsAnCViAhEgpJQKYC0yyAUAO7iQAsIRveXsjT4orACZchI8EVHISaEEvH0AZgEMArgBsALhKnKERWfMXKQqzboOS03OgqWCUMleu37DDpxddoJpbWXgacAUGuYAhKkBxmKBbcyKbOYhgUrvxGKSC8AJYAdgA6AA6sGUbRLjx5cHkgRYwVkBTJqUnc5kq8APZaBpVY3NKiOKkyvADyAKoAKvFEDihYpibyq2KcRFhKyCN8uyjC6a18BCumRBdoYD0gAOIAcgAii0f7vDt711vn8idvsdYi0KOtLpQwKlNrwAJJPNCgyhyFAHKipA5wp5sbYXFhKa7yGRxNTI+QHQmogR7aBIGCQRFTEDDXgkUzjEBsqBiUlEO4QoHITb85C0qCwRHbD4nFnA5RIylBSnLXgANQAwuqFR8LEC0CrHq9WAAnf7Is347moeCsUldOhgKGUDDgJ0isX0xFDA6WjmWvoDVhAA

pfalstad, 2024-05-02:
The top one that says "out" is not a labelled node, but an analog output.  Outputs and Labels->Analog Output.  It doesn't do anything.  You can optionally view its voltage instead of the text "out".  Unlike a labelled node, all the analog outputs are not connected to each other even if they all say "out".

The other two are both ordinary labelled nodes, but one is in all caps.

jjdnd, 2024-05-03:
thanks, useful to now since they behave differently, it stems from copy and pasting examples and not realizing they ere different component parts as they look very similar. it could be useful to make them more obviously different, maybe different colour text or something. but thank you for the info