pfalstad/circuitjs1 #88: Bug. Incorrect Diode behavior

inyutin-maxim, 2024-05-04:
I am draw scheme
[image]
7segment draw velue 3, but shold be 7

Similar sheme in EasyEDA (slightly simplified),

[image]

pfalstad, 2024-05-04:
Hi, when you post a circuit, could you post a link (File->Export As Link) rather than a screenshot.

The default Seven Segment LED element uses logic inputs by default, so if the input is at 5V the segment will light up.  It doesn't matter if there is any current, because the inputs don't draw any current.  In this circuit, there is no current since the ground is not connected, so all the connected inputs are at 5V, so they all light up.

You can right click the 7-Segment LED, click "Edit", and then change "Diodes" to "Common Cathode".  Then connect the ground, and then the segments will only light up if there is current.

inyutin-maxim, 2024-05-04:
Thanks Thanks, you right it was necessary to switch the seven-segment indicator to the mode on the common cathode. The problem was not in the place where I thought. This project is very cool

closed 2024-05-04 by inyutin-maxim