pfalstad/circuitjs1 #90: Comparotor OUT Spike

jjdnd, 2024-05-05:
In the following example OUT1 of the Comparator displayed to a graph has a blip in time with the Oscillator switchover

is this something i do not understand or some kind of bug.

other than the connection via VCC from NOT EAVEN the same voltage source as VCC2, i do not understand how the Oscillator can affect the Comparator output

[image]

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCAMB0l3BWEAWaA2ATARjWMmsEAOATg2RJDWRAHYQFIGBTAWiywCgB3EduIn2RN+kQYMg8+JNCOHSsGIUwycAbnywDl02TqZN5GaEgNQTnDJHqsSinayJE5hkACUAogDEMCKaPFNbVolSV5bPRCFJSjVDQC6JQimKJokLXgGKGyYPysbOxikpxSlGgAFAHUAWU4AcyCxEEF+cmacy2to9sdndpoAcQA5ABFOgvt2fD4SzWmaADUAYSX1RpasaeS5tByUJlMci3DCnozAsPWHc5mJfxkRTd3bSaeQVQAnIQ3IGlZkFryJgZOCcL4yII0CH-MwgyQAFxmZn+EmyWHMCBIYGQCAwtCwtBIRLQRAwlBgaEUVkgmKwyAw2JpIAyIAAJswAGYAQwArgAbeGcRGKVEQ2FCaBgSBoIlEijOQkkJCsMDoEjIRXCIgIWhoTHpOBszm8gX+AHM3YQnBQcbfHZ29h6RYrM0-KHPLQ0ST5O1YMh27C7QajW3sIhevhWVHO1Y+gm7FRNVyeLxaUOnN69YHzEAAZQAkkMPGsEpmHszpq5GO8TDXDrl7noy-Y3qpwjcoqXK-4RQ5zTD3lJ4zpoUYh7q0RhUW3NFPso4zKoAJa+rbmv2hcxwLg+l7FFqzXoodzeNPxG5H7ZHsxAtUQMwNk56S+nI8zhKXi9vS4o+2--g0FwAAeuiaNqKAeqSx5tEsAD2AC2AAOXIfLB8KwR8roWnaTiDuE5q4b+36cAAzmBEbsHOD4gNyfIkcwnAgawxBKGAlD-MgQZlO8NAAPIkQAxkufJ8ly6EfAAOiR75asyyB-Pggh0l6rpKfJQhgEoG42vhmnMv6KrhvpSjvggTYGQgLb+u+lK7MpfBmcC-o-mg1DvGZfCuYBeg-op7nPF5-k6Xwfn2SqJJBQiDnTDgci4bFaKeegvhEISIpEFgTjYpGsCpSQhKYFKpAUAggGGuy3L8oKu4AoIUbfHV8jBmM+HxROKJ1U0XBfB1ICjvF6nAvAoa1X1zyjW5MZgtFlrPAgMWDcy8Del0zHTNC827JNIDLKs4SbXJfwHQlkgCQ1h3KGpEYwMtkBwrdD2sdApCQHY820FKGAkvZN13ZwZ2sLZF1rXZi2-fA90PctJnQMpbEAvSmkYMj96wH94SBWFmM+aGgXQoF227dNgNufjbk-Utw27kD+PWoTLo9TTAXWhTIL+EDw6A9a9U7qtQP1VzuzkK4wxjIiJPeSIHM+cyGIkDSJJ2NiMoFaE6CcSQ4abHikC+OQEAshVJqCvhnZ9P8nbruz1oLp5NtW+LxB-LTzxirLuTy2ZpCbBqODIHg5jUDKWuabQuu0BHSnlcaVVCg5vygRbSRu+igPJcQaVTlOcBSsqxhiISqtfVoX1adHlWmtT9stEDjrHqLI2W4CnbWs1-12x6z5e5oMvg9uUMDzDcPqlrIcyKj27t8xVg93ILe92jEMD1DSSqi9b1gCEMgyvL9C-Vw+0JxEDkRY6Bj+E7ie4o8OOm0kCVJ5o9guTbD+1xwoTE7Xwsd5ozmU5IHq3dWA-2YqfbSQ1hpAJniA+Q09Hj-zhMTDUSRQEoL-puNmGMbagOAT-H819IxwPQbA8+d8MFCHnmQhyMDtLwNnsFMBGwLLgNvr-dgBl344yrs8UB39Ug7RdPtU+uCYH4MYg5KCIC9RCE4pGTA0E+KCWEqJcSUkD4OmIRsBMUhRxMGhNaFa9BOZGCaAOPiABVAAKrzGwvYBxnx0JYqxqhYJ9XoNGV6lBTCL1erqMyvg0ScDAIIMASBUx7BgHYnasEBRcjqAxIAA

pfalstad, 2024-05-05:
probably a roundoff error bug.

esaruoho, 2026-03-01:
As Paul acknowledged, this is likely a roundoff error. The comparator is built as an OpAmpElm + AnalogSwitchElm composite. During switching transitions, the high-gain op-amp (100k) can produce momentary voltage overshoots before Newton-Raphson converges, appearing as spikes in the scope. The fix would require tighter clamping in OpAmpElm's doStep() during saturation transitions, or adding damping during state changes.