pfalstad/circuitjs1 #101: 7 seg led relay always on in logic inputs mode when direct connect to relay

goldenshaw, 2024-08-11:
a and b segments both on
[image]

pfalstad, 2024-08-11:
The implementation of the relay connects a large resistor between the two terminals (the +5V and the a terminal) when the switch is off.  This means that the a terminal is still at +5V even though it can't conduct much current.  But in logic input mode, the 7 segment display doesn't need any current.  So you could add a pulldown resistor, or don't use logic input mode in the 7 segment display.  I could probably add pulldown resistors to the relay.

goldenshaw, 2024-08-17:
it is great if you could integrate a pulldown resistor inside, then we don't need to add it.
when a relay stand alone, both out endpoint show the color of high voltage, i think this is confusing.

esaruoho, 2026-03-01:
For anyone following this: @pfalstad's explanation above is spot on — the relay models the off-state as a large resistor (not infinite resistance), so the terminal voltage stays high even when the relay is off. The 7-seg display in logic input mode checks voltage only, not current, so it reads "on".

The three approaches discussed:
1. **Manual pulldown resistors** — works today but clutters the circuit
2. **Normal (analog) mode** on the 7-seg — current-sensitive, avoids the problem
3. **Internal pulldown resistors** in the relay element — cleanest long-term fix

Option 3 would involve adding a configurable pulldown resistance in RelayElm (similar to how MOSFETs model gate leakage). This would keep the off-state voltage near ground when no external circuit drives it high.

MatNieuw, 2026-03-01:
Configurable internal pulldown resistors in a relay, yes please. I am simulating circuits with large number of relays, and always have to use 1M pull down resistors.