pfalstad/circuitjs1 #99: 7 seg led grey segments not show in off state in white background mode

goldenshaw, 2024-08-11:
it will be better to show a very light grey, like that dark red in black mode
[image]

pfalstad, 2024-08-11:
I believe I tried this and it didn't look good.  The dark red is realistic in black background mode but there's no such thing as a LED with a white background.

goldenshaw, 2024-08-17:
like this, a light grey color for the off segment
[image]

fritzw, 2024-08-17:
There are indeed real 7 segment displays with a white or light gray background, e.g. http://betlux.com/led/0.80-inch-four-seven-segment-led-display-BL-Q80A-41,BL-Q80B-41.htmThis change would drastically improve readability.

pfalstad, 2024-08-18:
Well, ok but those are white LED's...  But I could try again making a very light gray color for the off segments.

closed 2026-02-17 by pfalstad