pfalstad/circuitjs1 #103: Subcircuits not working - Matrix error

yM4rvinz, 2024-09-08:
I'm trying to make logic gates and compact them on subcircuits. This is a NOT gate:

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWK0CcAOFYAskUDYBmArBBAdgTxBJCQUloFMBaMMAKAHdxK0QC0WHggBM-QVHYAlcHjj88fMHIZEqDBjlpQdMBOwBmPReKHL5AoUXDQkMSCMPCxWLEpXU3-MWFvIHXLLyOAzmmgRikOwA5kGaeGYebhqS3GCkLl7potTKqc65VNlilvlh1PLKERUa7CKQZLLVIbIJNdQgAIIA5HUNBVh56ZBCg1RCnexAA

When trying to use the subcircuit, the error 'matrix error' shows up:
[image]

I'm not sure why this is happening and what should I do about it.

pfalstad, 2024-09-08:
This happens because the voltage of the A input is undefined when the switch is open.  Normally the simulator detects this and fixes it by adding pulldown resistors, but it does a bad job when subcircuits are present.  Instead of using a switch on the input, you should use a logic input.  That ensures that the input voltage is always defined.

https://tinyurl.com/2yrmo5mv

yM4rvinz, 2024-09-08:
Thank you! I'm going to use these subcircuits a lot.

closed 2024-09-08 by yM4rvinz