pfalstad/circuitjs1 #105: [afilter, analog filter applet] "Options" > "Show Poles & Zeroes" not working

zach-lol, 2024-09-20:
Hello!

Sorry if this isnt the right place to put this. I am not very familiar with GitHub & I didnt see a repository for the analog filter applet (afilter).

Recently I have been using the Analog Filter Applet (falstadcom/afilter/). I have been playing around with it, trying to see what it is capable of. Something that caught my eye was "Options" > "Show Poles & Zeroes". When I click it, it appears as though nothing happens. All that I notice is that:
1) the entire circuit shifts to the left
2) the right hand side of the screen becomes unusable (cannot place or drag components over to it).
I guess my expectation would be that the bode plot on the bottom would be marked with X'es and O's for poles & zeros respectively.

The only way I could get the feature to work is:
1) clicking the initial link you get from googling "falstad analog filter", so that the starter RC circuit with the cap having the "cutoff frequency" slider.
2) When "Options" > "Show Poles & Zeroes" is clicked, then a plot shows up on the right, and a line traces over it when hovering over the bottom bode plot.
3) If I try to add another capacitor to the circuit, it breaks, and I can get neither the right hand side plot nor the bottom right hand side "Poles: _" info to show up again

Questions:
*Not sure what the plot on the right is, what its units are, or how to read it.
*Not sure why it only works when that one cap with the slider is in the circuit.
*Is there something I am missing?

pfalstad, 2024-10-06:
I answered this already in private email but I'll put my answer here for future reference..

I never really finished the analog filter applet.  It doesn't calculate the poles and zeros for an arbitrary circuit.  It only works for the predefined filters.

GilDev, 2025-03-09:
Didn’t even know about that applet, it’s great! I’ve been looking for a while to get such a dynamic Bode plot for my students. Too bad it’s not finished yet but thank you for what has already been done! Looking forward to try out any improvement on this side.

esaruoho, 2026-03-01:
Paul has clarified that the analog filter applet's "Show Poles & Zeroes" feature was never fully implemented — it only works with predefined filter circuits, not arbitrary user circuits. This is a known limitation of the afilter applet. Since this is acknowledged by the maintainer as unfinished rather than broken, this issue could potentially be closed as "won't fix" / by design.

pfalstad, 2026-04-14:
It shows poles and zeroes of most arbitrary circuits now.  If they are too large, or if they contain transmission lines, we can't obtain the poles and zeroes, but we display a message instead of just a blank area.

closed 2026-04-14 by pfalstad