pfalstad/circuitjs1 #114: Details on the 3-phase motor

JonasForssell, 2024-10-18:
Just found the motor element. Very nice indeed.

But is there any more documentation on how it is implemented somewhere?

Is it a Pm motor? What does the coupling factor do?

Could the resolver angle be available on a pin? That would open up for some nice control modelling.

Many thanks for this program. I use it extensively!
/Jonas

pfalstad, 2024-10-20:
It's based on this model.

https://forum.kicad.info/t/ac-motors-simulation-1-phase-3-phase/14188/5

which is based on this paper (in French): https://hal.science/hal-01626183v1

No, it's not a permanent magnet motor.

esaruoho, 2026-03-01:
Paul provided references to the implementation sources: based on a KiCad forum discussion and a French academic paper (https://hal.science/hal-01626183v1). The motor is NOT a permanent magnet type. The coupling factor parameter and resolver angle output pin suggestions remain open as potential enhancements. Recent commit bf141c9 fixed the 3-phase motor resetting currents on stamp.