pfalstad/circuitjs1 #120: bug when using dpdt switch

LE1ED4523, 2024-11-06:
[https://falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKACcLiURttDPuAFmzcqYeOxAZIgnnykyQw0cjgtBKYiEIYhhWTu6F8UcCABmAQwA2AZzogULAO4ClI7SlnKoLz949CLyV9X1dpAKNdd1E-QxjtaMFQyDiTYzxtdMEqVNcM7RztBH59XLiSwqpCSoK8qqzMsu08TPrCVsaWpuj2zviEFB7Y10HhiiG5fnaijsyMDu6wqUW5lczeaZYUMH4ELh5FCMPZXNhWUcnNt2v6hY35Y9u-fe5r15Plp6PFZ6A](url)
this circuit use a dpdt switch to switching between series and parallel. but not work when toggle the switch

pfalstad, 2024-11-06:
Hi,

The code that detects a valid current path through the ohmmeter is not handling the DPDT switch properly.  It looks tricky to fix.

For now, try this: https://tinyurl.com/24bymdh8

or this (since tinyurl is acting weird): https://falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKACcLiURttDPuAFmzcqYeOxAZIgnnykyQw0cjgtBKYiEIYhhWTu6F8UcCABmAQwA2AZzogULAO4ClI7SlnKoLz949CLyV9X1dpAKNdd1E-QxjtaMFQyDiTYzxtdMEqVNcM7RztBH59XLiSwqpCSoK8qqzMsu08TPrCVsaWpuj2zviEFB7Y10HhiiG5fnaijsyMDu6wqUW5lczeaZYUMH4ELh5FCMPZXNhWUcnNt2v6hY35Y9u-fe5r15Plp6PFZ9GD3CyD5ecocYFFY4g0zieAScKKKHfU4sIA

pfalstad, 2024-11-06:
You could also use two separate (but linked) SPDT switches like so:

https://falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKACcLiURttDPuAFmzcqYeOxAZIgnnykyQw0cjgsA7gKUiQhFLOVQNu-du56DhWZGPSDOwhiE6bmx8-NOlVo2-y7-QkDBKlcAvF0Q3QR+K1DjQhjIqkTY-zC48OTdPAiM3KzCAvdfHIiShBRyrzDK6u46uX4MqKKIjCKy0o7ygp6moxQwfgQuHkU7cesoWFZNRt4RscXuzpXJldrl+VHuXGtbRX2FWU2WAGUTS1kLLNCQADMAQwAbAGc6cBAUC90vQR8JSCeSgj1eHy+PyAA

fritzw, 2024-12-10:
> [https://falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKACcLiURttDPuAFmzcqYeOxAZIgnnykyQw0cjgtBKYiEIYhhWTu6F8UcCABmAQwA2AZzogULAO4ClI7SlnKoLz949CLyV9X1dpAKNd [...]

The circuit from the first post works fine for me on Safari mobile (iOS 17.6.1). The ohmmeter switches back and forth between 500 Ohms and 2k Ohms.

pfalstad, 2024-12-10:
> Clicking on an SPDT doesn't toggle it for me anymore. Double clicking it opens it config panel. The "make before break switch" is anything but.... Acts like a tristate switch. Your site doesn't even open up sometimes recently.... What have you done bro?!?

Works for me.  What platform?  What browser?  Which circuit?  It sounds like you're talking about mobile, since you say double clicking it opens the config panel.

Yeah, the "make before break" switch has three states.  There's one state where it's connected to both poles.

When the site doesn't open up, what do you see instead?  Is there anything in the console?

pfalstad, 2024-12-11:
Not in a circuit?  You mean a new circuit?

When the simulator loads, I can just select "Add SPDT Switch" from the right mouse menu, create one, click on it, and it works fine.  Can you do that?

So you see "Can't load circuit!" sometimes?  Does this happen a lot?

pfalstad, 2024-12-12:
I might have to think about switching providers.

closed 2025-02-26 by pfalstad