pfalstad/circuitjs1 #122: How to input customized current signals?

BunnyIcon, 2024-11-08:
A very good software. I want to know how to input customized current signals? What is the "Add Data Input" format? What is the default unit?

pfalstad, 2024-11-08:
"Add Data Input" is a voltage source that varies over time, using an input file to specify the voltage values.  The input file is a text file with one value per line, specifying the voltage in volts.

You want a current source that acts the same way?  Use a VCCS to convert the voltage values to current values (in amps).  Example: https://tinyurl.com/2a67qmne

BunnyIcon, 2024-11-08:
Thank you very much. That’s a really cool simulation software. It helps me a lot.

Best Regards,
Jiayue Zhang

Tsinghua University, Beijing, P. R. China
Post code: 100084
***@***.***
Tel: +86 188-1002-8806

closed 2024-11-11 by pfalstad