pfalstad/circuitjs1 #124: Convergence failed

Olar55, 2024-11-20:
Hi
Please find an issue on circuit with message convergence failed

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWK0CcAWAHAdiwZgRrgGz4YIBMRIaaISCkdApgLRhgBQutAWkwHZMATgB0Azrmi5xANSjhoWMGnzkMYFGrDxyrLPPIgiUqBwBeIVZeKWKIcuVq4QfQaIlTZHAOa3DuGytiDHlIDgB3OjVwTWpMGMMwoSiQ8ixDIkd7dPlteCS-Iyc7B1pGMAYOADcQLCIQgKo61IQqRkY0crBoBCR25AQIwsbLDCcbMMi0-0C7EcnsjOKl006IClTIEMzaci3wEAxxCvE2BGP8uHFjbjBCe4xyAIRcLHEEaDQsOFwMSE0sCgwPgiEQLvkoOI9uDLvBwOICjRtrQkQl5F84EMdvZ9tjpqYpjkRhoZm0OGAsCj4iTqEQnHsDnk4MgiHcQgAlJhiACWYgALgBDfgAYyYQ1RI0wjHmQwQYFJRiJEyGVjpcWRpgpISwYyKtV1yloVAgMESsGIFMgmTSuDU5AoKGQkCQTLhMEghjyToqTud4CGOqNtAwKCoaoWRByNPS5ViCzlK2x4fFf3srWoqZlkQT1ioUtzBJS9iyG0WhdLpVGuyyYQAJorPbFIytDLWmAAzAUAVwANnysTk1TG9WEajTibFDaFqOVUG1Qj0OPWaVOaWrWx3u33xfFbR1d8rIs001R1FR7WTImeCyGqDKAA6HMDn9PXkbOdocR-49dKu+hDg9n0W89RAik8xAABlAB7IQ+W5JhxAAFQASQAWSGECRhAi9C2PbDnwLBZAz1LA7GTPkjCyPZthyGj5DYZBlEcYhIHwVpn1aR4QBYchoHIDQKCtf4yMgOoiBdZk207Xt+2SJMjWo-ZpXgLEsjVZsRwDOwpzIxtEgDepwBoWojPoiM6PUv9C00kZ8MPQpK1+asyh3BpAl1B8qxPbz3wAyI0HXdN81whZUTVHNk0JFY11cx8c1wnM-M-bNohpBL0wKY96OyzLwDCZ9DGHIcGSnDcZL7FgeyYZdp3dThkmHXTSrjfLMSPBkh3I1yjx02ImrjDhoPkcMZxQR1tCdPp7HkZwzAEYQhroEdwHwCkkAZd0XRmpBcCW9gVs6cacV9SBBNweUNpmg7nAAeS7PlxGFaCAFt7yWlYOn+Cb2lgeAUDqVp7UsGb-BAABRfg+SEJhELEAUe25F6uBCJAADETVyJBGJkaC+wFbwxV+bIQAx6dlB4iBcfxwmOCAA

pfalstad, 2024-11-20:
?  I tried it and I don't get a "Convergence failed" message...

Forecaster, 2024-12-06:
I see the `Convergence failed!` message in the bottom left and the simulation is unable to run:

[image]

Olar55, 2024-12-07:
And did you find the reason?
Is it a bug or a problem in the circuit ?

pfalstad, 2024-12-08:
Ok I see it on windows.

I put a delay buffer after the comparator and that seems to fix it.  I'm still not sure if the circuit is working properly though.

https://tinyurl.com/29pzewaw

The comparator in this simulator is infinitely fast which is not very realistic and can cause convergence issues.