pfalstad/circuitjs1 #128: Netlist Support

JK-c66, 2024-12-12:
Hi, is there any netlist support in this simulator or are there any plans for it

I saw the current one which has the text export and import but I didn't understand the format, 
it would be nice if the importing and exporting were from something standard as Netlist for example

pfalstad, 2026-10-18:
Apparently converting to KiCad directly wouldn't be too hard.  You want to generate PCBs for your relay circuits?

eabase, 2026-10-20:
> converting to KiCad directly

Sounds awesome, but sometime one want to edit the *text* files directly, especially when it comes to *search/replace* operations for large quantities of similar components.

pfalstad, 2026-10-21:
>Sounds awesome, but sometime one want to edit the _text_ files directly, especially when it comes to _search/replace_ operations for large quantities of similar components.

Sure, but for that you should just edit the native xml format, right?

MatNieuw, 2026-10-21:
Exporting (via "save as" maybe even)  to KiCad schematic compatible format would be great! I indeed want to make PCBs for most of my relays circuits, with over 100 relays/pcb. Prototype boards, manually wired would work too, but very much more work, and worse, any wiring mistakes would be hard to find . If Kicad doesn´t know a particular component is no big deal, you can fairly easily create custom components.
I do indeed edit text files directly sometime, just yesterday I consolidated several different relay types in a subcircuit with 120 relays to one standard type. Just a few minutes search-and-replace, instead of very tedious clicking each of the relays and change the model, and the check the text file afterwards to see if you did not miss one.

eabase, 2026-10-21:
@pfalstad 
Hi Paul, 
 
> Sure, but for that you should just edit the native xml format, right?

TBH. I don't know. I have not used this tool for at least a year, since my last comment before. So I don't remember how it works exactly. But I do remember wanting to edit multiple components (and wishing for a *netlist*) at some point, and it was a terrible PITA without plain text files. So using XML is not a human solution. But anytime I need to check some circuits this tool is the first place I come to. 

Sorry, I couldn't provide a more useful comment.

MatNieuw, 2026-10-21:
Oops, just a netlist would be fine for making PCBs. No need to recreate the schematic in Kicad first.
@eabase , the xml format is very simple, much like the earlier text files. Not like the typical xml you see. Editing components I find easy.

pfalstad, 2026-10-30:
ok added kicad and spice netlist support to the dev-ts branch.

https://pfalstad.github.io/circuitjs1/circuitjs.html

look in the File menu.

MatNieuw, 2026-10-31:
it works! Great, this will save me tons of time (mistakes...). I tried with a large and small circuit, seems OK; I need to verify lines to be sure, before I go into PCB design.
The spacing of the schematic is larger than I normally use. Is is possible to reduce spaces between horizontal and vertical parallel traces to half to what it is now? Or allow choosing between wide/narrow on export? Unless here is a standard/convention of sorts for this.

pfalstad, 2026-10-31:
the kicad spacing is based on the spacing of your original circuit.  I tried reducing the scale but it didn't work out.  so the circuit would have to be compacted which is more complicated.