pfalstad/circuitjs1 #132: Feature Request: Loss free resistor

EmergencyTemporalShift, 2024-12-20:
https://en.wikipedia.org/wiki/Loss_free_resistor
Loss free resistors are a type of resistor that feed power back into their source instead of dissipating it as heat. An example circuit and internals would also be nice to have.

pfalstad, 2024-12-20:
You filed an issue about this already: https://github.com/sharpie7/circuitjs1/issues/555

I used a VCCS to model it: https://tinyurl.com/yanxq2z5

EmergencyTemporalShift, 2024-12-20:
My bad, I thought I did, but forgot there were two forks when I didn't find it.

EmergencyTemporalShift, 2024-12-20:
I still think it deserves a place in the index with a little blurb though, it's educational.

EmergencyTemporalShift, 2024-12-20:
> I used a VCCS to model it: https://tinyurl.com/yanxq2z5
Is the capacitor supposed to gain more charge over time? Is it because there is no real load? I might just not understand the concept well enough.

pfalstad, 2024-12-20:
I think so.  The capacitor stores all the energy that was drained from the source.

EmergencyTemporalShift, 2024-12-20:
And how do you tune the "resistance"?

pfalstad, 2024-12-21:
The output function of each controlled source includes a 1000 in the formula..  Change this to the resistance you want

EmergencyTemporalShift, 2024-12-21:
I'm trying to make the resistance adjustable, but I just get a singularity error. https://tinyurl.com/2alykrwt Is this a problem with the implementation or my circuit?

pfalstad, 2024-12-21:
replace c with max(c, 1) in the first controlled source and replace e with max(e, 1) in the second.  The problem is that c/e are zero in the first iteration, so the current is infinite.

pfalstad, 2024-12-21:
It looks like you're trying to use that energy for something, which won't work with a VCVS output.  This works better:

https://tinyurl.com/25hw4gt5

esaruoho, 2026-03-01:
Paul's VCCS-based solution in the comments above already provides a working Loss Free Resistor model using existing elements. It might be useful to add this as an example circuit in the circuit library for discoverability, since it demonstrates an important power electronics concept that can be built from existing components.