pfalstad/circuitjs1 #134: [BUG] Shared Labeled Nodes Truncated to One Pin when Creating Subcircuits

SvarunSoda, 2024-12-22:
Attempting to create a subcircuit from a circuit which has more than one shared labeled node always results in the shared label node groupings to get truncated to a single pin on the subcircuit.

Specific example:
[image]
[TEST_1.txt](https://github.com/user-attachments/files/18223541/TEST_1.txt)

In the circuit above;
- the `GND_FIRST` and `GND_NEXT` labeled nodes get truncated into a single `GND_FIRST` node
- the `3.3V_FIRST` and `3.3V_NEXT` labeled nodes get truncated into a single `3.3V_NEXT` node
- the `OUT` and `DIV_NEXT` labeled nodes get truncated into a single `DIV_NEXT` node

The subcircuit creation script appears to automatically identify & truncate any shared labeled node groupings to a single node, which is not the intuitive or expected behaviour here. In my specific case the shared labeled nodes all serve unique purposes despite being shared, and I am unable to create a subcircuit because of this feature. At least this feature should be a toggleable option.

fritzw, 2024-12-23:
Does it help if you put a zero ohm resistor between the two nodes?

pfalstad, 2024-12-23:
You can't have a zero ohm resistor but a one ohm resistor should work.

SvarunSoda, 2025-01-15:
Thanks for the comments;
putting a 1 nano-Ohm resistor does the trick there, albeit this should still ideally be a separate feature.

pfalstad, 2025-01-15:
re-opening since it should still be fixed.

esaruoho, 2026-03-02:
PR #264 fixes this by changing the deduplication key in `getCircuitAsComposite()` from node number to label text, so that different-named labels on the same electrical node each get their own pin on the subcircuit chip (like real ICs with multiple GND/VCC pins). Also removes the "Can't have two input/output nodes connected" check that blocked model creation in this case.