pfalstad/circuitjs1 #135: Bad connection reported, version 3.1.1js (online 2024-12-29)

MatNieuw, 2024-12-29:
A simple circuit won't run using straight connections to the bottom GND line, it does when the connections are to the left GND line, which is the same circuit. circuitjs reports 'bad connections'. If the bottom of the relay coil and of the 1k5 R are connected to the left vertical line (GND and minus of the 150u C) it works.

Circuit:
===========
$ 1 0.00009999999999999999 1.700203994009402 54 24 50 5e-11
178 240 160 240 272 6 2 2.475 0 0.05 10000000 0.016 700 0.008 0.01 0
r 224 272 224 352 0 1500
r 256 272 256 352 0 470
209 144 176 144 240 0 0.00015 0.001 0.001 100
w 240 160 208 160 0
w 208 160 144 160 0
w 144 160 144 176 0
w 144 272 144 240 0
w 144 384 304 384 0
w 304 272 304 384 0
w 256 352 272 352 0
w 272 272 272 352 0
w 272 352 384 352 0
v 416 384 416 320 0 0 40 24 0 0 0.5
w 384 352 384 320 0
w 384 320 416 320 0
w 416 384 304 384 0
w 288 160 384 160 2
r 464 304 464 384 0 470
w 416 384 464 384 0
w 464 304 464 160 0
w 464 160 384 160 0
g 416 384 416 400 0 0
w 144 272 144 352 0
w 144 352 144 384 0
x 385 306 413 309 4 14 24\sV
x 319 208 446 211 4 14 relais\s24\sV,\s700\sohm
x 319 225 438 228 4 14 on\s16\smA,\soff\s8\smA
w 224 352 224 384 0
w 208 272 208 384 0
===========

MatNieuw, 2024-12-29:
Sorry, main post had bad format, and cannot change it.

[image]
[relay_clock_generator_bad.txt](https://github.com/user-attachments/files/18269127/relay_clock_generator_bad.txt)

pfalstad, 2024-12-29:
You can't make connections to the middle of a wire.  You can only connect to wires at their endpoints.

MatNieuw, 2024-12-30:
Indeed, that made it work. Thank you.
Strangely, if I delete part of the bottom line between left (below the 150u C) to the connection left of the voltage source, then added the line sections one by one from right to left, it still gave 'bad connections' . When I delete all of the bottom line left of the voltage source, then added line sections from the left to the right, it was OK.
Anyway, now that I know it, it's OK.
I am wondering though, if the vertical line terminate on the horizontal, and circuitjs reports a bad connection, what is the philosophy of not making the actual connection there and then?