pfalstad/circuitjs1 #136: [QUESTION] how the demonstrations display graphics?

Adomo-Software, 2025-01-05:
Hello, I'm wondering can these demonstrations display/draw graphics, either character or pixel based: 
https://www.falstad.com/pong/net.html
https://www.falstad.com/pet2001/
Is the source code not available?
I would literally pay for an explanation or source code, making basic video game graphics on the bare-bones circuit sounds very cool!

pfalstad, 2025-01-05:
They use the js interface.

https://www.falstad.com/circuit/jsinterface.html

It uses callbacks to get the state of the circuit so it can read the video output.  The code to draw the screen is in javascript.  You can just look at the source code for those pages to see how the drawing is done.  It's semi-readable.

Those pages use the digital branch of the circuit simulator because it's faster for digital circuits: https://github.com/pfalstad/circuitjs1/tree/digital

But you could do the same thing in the main branch.

Adomo-Software, 2025-01-05:
Hi, very thanks for fast response! :D

That makes sense, you can write javacript to read values from output circuit components, and then display those values in <canvas> element.
i still am confused where to inject that javascript code.

pfalstad, 2025-01-06:
Yeah, you can embed the <iframe> in your page.  But you need the simulator to be hosted from the same page as your code.  Otherwise you won't be allowed to access anything.

More info here:

https://www.falstad.com/circuit/doc/js-interface.html

Adomo-Software, 2025-01-06:
Much appreciated.

closed 2025-01-06 by Adomo-Software