pfalstad/circuitjs1 #137: Diode not dioing

Lipino777, 2025-01-08:
I think the picture is selfexplenatory:
[image]
[WhatInTheDiode.txt](https://github.com/user-attachments/files/18348204/WhatInTheDiode.txt)
 it just goes trough. i know diodes IRL can let some tiny ammount of current trough, but I assume this default diode is supposed to be a theorethically perfect diode

fritzw, 2025-01-08:
Yout assumption is wrong, the default diode model is not an ideal diode. See also #88.

But I agree that combining logic inputs (which draw no current) with diodes leads to confusing behavior. Maybe there should be a model for an ideal diode?

Lipino777, 2025-01-08:
Greetings,

i see now. Thank you for the clarification and sorry for the
misunderstanding.

Do i delete this issue now? I am sort of new to this kind of thing. Also
thank you for taking time to review this.

Have a nice day/evening.

-Lipino

pfalstad, 2025-01-09:
The diode is not ideal.  But I think the bigger issue is that the counter inputs ARE ideal.  They have infinite input impedance.  So they draw no current at all; they only look at the input voltage.  Since there's no current across the diode, there's no voltage difference either.  A pulldown resistor would take care of the problem.

[image]

closed 2025-02-26 by pfalstad