pfalstad/circuitjs1 #144: LM317 functional block is not working

motla, 2025-02-07:
LM317 equation is Vout = 1.25 x (1 + R2/R1)
The LM317 functional block inside circuitjs is unable to regulate Vadj to 1.25 V
With R1 = R2, the output should always be 2.5 V

Here a minimal example with R1 = R2 = 10kΩ
[image]
https://tinyurl.com/27qoan2x

The model must thus be corrected or removed.

pfalstad, 2025-02-07:
The datasheet says:

"8.4.3 Operation at Light Loads
The device passes its bias current to the OUTPUT pin. The load or feedback must consume this minimum current for regulation or the output may be too high. See the Electrical Characteristics table for the minimum load current needed to maintain regulation."

The Electrical Characteristics table says that the minimum load current to maintain regulation is 3.5 mA.

You have no load current here.  Connect it to a load resistor smaller than 1k or so.

motla, 2025-02-07:
Oh ok thank you, sorry you're right! I'm used to more recent LDOs which doesn't have these limitations so I didn't had that in mind.

closed 2025-02-07 by motla

motla, 2025-02-07:
By the way, many thanks for your work, this is a really great tool! 😉