pfalstad/circuitjs1 #146: Transformer current not correct

HansRamon, 2025-02-12:
I have noticed an issue with the transformer component in the simulator. Currently, the voltage and current behavior at the primary and secondary sides do not align with the expected power balance.

Normally, the power should be equal on both sides, following P = U × I. This means that if the voltage is higher on the primary side, the current should be lower, and vice versa. However, in the current simulation, both voltage and current appear to be higher on the primary side, leading to an incorrect power calculation.

Could you please check this issue? Let me know if you need any further details.

link to an example:
https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3AWAnC1b0DYRNpMB2MAZmLgFYAmMADgjIRDMckYFMBaMMAKADdx8WCvgoCsYBAxZ0WRCkxaLoZbgHcQRSiAoUGm0cNGRuAFW0jwSUfstHwcbSHYxIZMkUgIdXCfnzUSKFh5QgoSQjk5MHpqINQ1c1EwK0TwSSgEw21dVKzjdQIhCyy8hJssyVjSgCcQSrSGerz7eDKtZOt29PzGdIlGhFj+jPV6BgrB1OMAB16GYbHUonAoFvg4blnpbqWV6XXW9W35wQaR1azCqe4ifGlT4cv05YhjW9kugaHnve4gA

Best regards,
Hans

pfalstad, 2025-02-12:
I'm seeing that the voltage on the primary side is half of the secondary voltage, and current is twice the secondary current.

229.8 * .0928 = 21.325 W
459 * .0459 = 21.068 W

seems about right...

HansRamon, 2025-02-13:
In follow case it is not good with a ratio >=1 (in the example 5)

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3AWAnC1b0DYrQwJgKwIDscS+YSAzBpZSIffZPQKYC0YYAUAG7hFZcRXPyxgECKOCYhKBKcxj4uAd1n4RuXJMoaQQkZC4AVfcPBIRukRUPg42EWxhwM+JAkoAOL7jAkESDciEGdYSFxaYQwiL0skDC8abFRVMxtLdPAJKDSDfW0s-KM1f0FzfOK06yyJLyyjACcQOuzJVuL7eGq9W3UbHJL6HPF2hHrR3LVCSXyO8yMAB2HJSZmsughmMHhdrmXtwY3pbd3utUPVgTapqXyyhq5KElEb+5zN3OfmGsnWyc+RiAA

ratio 5
229,81*0.013 = 2,99 W
45,9*0.004596 = 0,2109564 W

ratio 10
229.8*12.6 = 2.9W
22.958*0.002298 = 0,06854

pfalstad, 2025-02-13:
The circuit never reaches steady-state because there is no resistance in the primary.

I changed the phase offset of the AC source to 90 degrees so that it starts closer to steady state.

Also, I increased the inductance, which brings the transformer closer to ideal.

https://tinyurl.com/22hfwcgu

HansRamon, 2025-02-13:
Thanks for your answer. With This diagram I can work with the students. Maybe put the small resinstance as a parameter of the transformer because a coil has always a resistance.

Thanks for the help. I'm a big fan of the simulator by the way.