pfalstad/circuitjs1 #152: Bug- Unexpected potentiometer behaviour

the-electronic-man, 2025-03-03:
I have a simple operational amplifier set up as a non-inverting amplifier:

[image]

And I want to fine tune the gain with a trimmer potentiometer to get a peak-to-peak voltage of 10V, but when I add it in the feedback loop it lowers the amplitude and distorts it.

[image]

If I go back to the first circuit and increase the resistance of the simple resistor, the amplifier works as expected, amplifying the signal:

[image]

The operational amplifier has enough voltage room as it has these settings:

[image]

Please let me know if it's just something wrong with my circuit, but I think it's pretty clean.
Thank you!

Joerg-rw, 2025-03-04:
seems your o'scope shows the voltage across the 22.9kΩ resistor. Maybe add a voltage probe for the o'scope to the OPAMP output

pfalstad, 2025-03-04:
hi could you send me the circuit using File->Export As Link or File->Export As Text ?

the-electronic-man, 2025-03-04:
Hello, this is the file, but you are right, it was the voltage across the 22.9kOhm resistor. I always thought the creating a scope from a node gives me scope probing the voltage between that node and the ground but I was wrong apparently.

[image]

$ 1 0.000005 2.922428378123494 50 5 43 5e-11
a 416 288 496 288 9 15 -15 1000000 0.9983653539204881 0.9984717873334794 100000
r 496 288 496 368 0 22900
174 496 368 496 448 1 100000 0.7376 Resistance
w 512 416 528 416 0
w 496 464 528 464 0
w 528 416 528 464 0
r 496 464 496 544 0 10000
g 496 544 496 560 0 0
w 496 464 416 464 0
w 416 304 416 464 0
R 416 272 352 272 0 1 1000 1 0 0 0.5
p 576 368 576 288 3 0 0 10000000
w 576 288 496 288 0
w 576 368 496 368 0
o 1 2 0 4099 5 0.00078125 0 2 1 3
o 11 2 0 4098 5 0.1 1 1
o 4 2 0 4098 5 0.00078125 2 2 4 3
o 12 2 0 4099 20 0.00009765625 3 2 12 3

closed 2026-02-17 by pfalstad