pfalstad/circuitjs1 #171: Display somewhere (heading?) the name that the project was last saved as, or loaded from.

brewmanz, 2025-06-24:
I just came back to some work from yesterday and ... I don't know what I'm looking at. 

Heading line states blandly 'Circuit JS1'

'File' - 'Save As' doesn't help me
'File' - 'Save' just goves me a '(saved)' in bottom right panel

I'm developing twin versions (with two ideas, to see which one works better) and ... is it V1 or V2?
It would be REALLY GOOD to display what filename it was saved as, or loaded from.
Maybe it does display it somewhere, but ... if so, where?

Some programmes have a 'File' - 'Properties' option that helps in this situation.

'File' - 'About' shows 'Circuit Simulator version 3.1.3js.' on top line

PS it also displays 'snow' for 'Simulation Speed', 'Current Speed', & 'Power Brightness' bars after re-activating the programme.

MatNieuw, 2025-09-09:
I noticed that if I start circuitjs from the file manager, with the circuit name on the cmd line, the file name is not shown. But if you use File->FileOpen to open a file, the first about 25 characters of the file name are shown under "Current Circuit" .

MatNieuw, 2026-03-22:
If the above can be made to work, could a cmd line parameter like --help made to show all available cmd line options?